FYSETC/FYSETC-E4 #1: FYSETC-E4 Setup Example For a Delta Printer

Loadmaster1, 2020-12-04:
The provided E4 is for a cartesian printer.  Is there a chance of a working example for a Delta type of printer?  I have a Tevo Little Monster that I am trying to use the E4 with.  I keep having issues with the build failing.  Something to do with the Pins.  

I will continue to try and resolve the build error, but any input would be much appreciated.

GerogeFu, 2020-12-04:
Sorry , this board only have 3 endstops pin. And there is no enough pin for a servo or bltouch.  But i have a thought , if you don't have a heatbed , maybe you can use the heatbed pin PC9 for bltouch. That means , you need to change pins_FYSETC_E4.h file as below.

Loadmaster1, 2020-12-04:
Geroge,  thank you for your suggestion.  I will give it a go and see if I can figure this out.

Loadmaster1, 2020-12-04:
Geroge, I really like the design and wifii of the E4 board.  Does FYSETC have any plans to build a board like the E4 with added endstops for use with a Delta design?  I believe there is a huge market for such.  It would be worth considering.

GerogeFu, 2020-12-05:
Yes , we are planning now. Hope more and more people like E4 boards. And it is nealy support Grbl_ESP32 firmware now. https://github.com/bdring/Grbl_Esp32

Loadmaster1, 2020-12-05:
That is great news.  You guys did a nice job on the E4.  Yes, I'm sure there will be a lot of interest.