FYSETC/FYSETC-E4 #8: Fan always on

sdavidozzo, 2020-12-25:
I installed the firmware found here on my ender 3 but the fan is always on.
Sending the M106 S0 command or the M107 command gives me a message:
Ok
ok
but the fan remains on as well as the LED on the board.

GerogeFu, 2021-01-06:
Can you take a pic for me? It is a bit hard to tell from what you describe.

sdavidozzo, 2021-01-13:
[image: E4.jpg]
When I turn on the printer, the fan for cooling the filament turns on as
well as the red LED. When printing begins, the fan stays on and in this way
the piece does not stick to the plate. It never goes out even when sending
the M106 S0 command

GerogeFu, 2021-01-27:
Are you using stock firmware ? Or you compile the firmware yourself ?

sdavidozzo, 2021-01-27:
I tried both with stock firmware and compiled firmware but can’t solve the problem

Inviato da iPad

GerogeFu, 2021-02-02:
Do you have a multimeter? You can measure the vol of mosfet pin1 , if the voltage changes if you send commands(M106) to turn it on then off. Then the mosfet is dead. If the voltage of pin1 not change, that is the firmware or other issue not mosfet. 
[image]

sdavidozzo, 2021-02-02:
The volt of mosfet pin1 change from 0 (M106 S0) to 1,84 volt (M106 S200) . The mosfet is dead?

GerogeFu, 2021-02-03:
Can you measure this pin vol for me ? 
[image]

GerogeFu, 2021-02-04:
No matter how, contact our sales to send you another E4 board for you.

iNDeX999, 2021-08-06:
I had the same problem, Mosfet dead, new board is on it's way.

sdavidozzo, 2021-08-07:
I had to reinstall the original card and am still waiting for the replacement. I have lost all hope of seeing this card working by now.