FYSETC/FYSETC-E4 #12: Servo Usage, which pin is accessible?

theomega, 2021-03-24:
I want to use a servo instead of the Z-Axis Stepper motor. The servo would be PWM controlled which is something the ESP32 firmware can support on every GPIO pin.

Is there any GPIO pin from the ESP32 accessible on the board that can be used for this?

Tigger777, 2021-04-21:
I used GPIO0 for a Z-probe ABL servo. You need to solder a wire to the ESP32 (bottom pin on the RHS). You also need a pull-up to 3v3 else it won't always boot.  I had already installed everything so I just tied to to 5V with a 39k resistor. 

[image]

lols, 2021-06-16:
you can use the z-min plug for servo but you need to see the board pcb 
-remove the opto isolator + filter cap + 3.3v pull up resistor(red box)
-bridge the opto isolator pin(green line)
[image]
[image]

car-par, 2024-10-20:
Hi, I'm new to this and I would like to use a BLTouch on a Fysetc E4 but idk where to connect the PWM cable... and I didn't really understand what you all said in the comments above... Please could someone help me?

Tigger777, 2024-10-21:
> Hi, I'm new to this and I would like to use a BLTouch on a Fysetc E4 but idk where to connect the PWM cable... and I didn't really understand what you all said in the comments above... Please could someone help me?

Look above at my screenshot, the orange arrow is pointing to GPIO0, thats where to connect the PWM signal.
Can you build the Marlin code? This has been quite an undertaking for me every time there have been updates.

car-par, 2024-10-28:
Hi, sorry if I'm replying after a week, I didn't see you had replied. Ok I understood that I had to connect the PWM to GPIO0 but as for the rest it's not clear to me how to proceed. And yes I can build the code. Thanks for your reply