FYSETC/FYSETC-E4 #25: 'class CardReader' has no member named 'diskIODriver'

brunoeiti, 2021-08-28:
.pio/libdeps/fysetc_e4/ESP3DLib/src/sd_ESP32.cpp:324:30: error: 'class CardReader' has no member named 'diskIODriver'

Latest master branch have build problem caused by dependency change

brunoeiti, 2021-08-28:
I could build and "solve" the problem editing the sd_ESP32.cpp on openDir function that was failing to build
changing the code:
   if (!sd_volume.init(card.diskIODriver())) {
        return false;
    }
To:
    if (!sd_volume.init(&card.getSd2Card() )) {
        return false;
    }

zbyszp, 2021-10-30:
I have the same problem, any solution?

zbyszp, 2022-02-22:
> I could build and "solve" the problem editing the sd_ESP32.cpp on openDir function that was failing to build changing the code: if (!sd_volume.init(card.diskIODriver())) { return false; } To: if (!sd_volume.init(&card.getSd2Card() )) { return false; }

works for me