FYSETC/FYSETC-E4 #37: SD Card Not Detecting

lancesand, 2022-05-20:
Hello, I recently received one of these boards but I'm having trouble using the SD card on it.  It does not seem to be detecting an SD card present in the slot. Has anyone else dealt with this error?

marcioscardua, 2022-09-12:
I have same problem

NellClarke, 2023-06-27:
Has this problem been solved?
I can't find any solution for it, and FYSETC doesn't seem to update the code at all anymore.

crodenas, 2026-08-19:
I know I'm late to the party but was having the same problem recently with my new board.  

Someone smatter than me will know why, but from experience I can say top card did not work but the bottom does work, both brand new.

[image]