FYSETC/FYSETC-E4 #40: Failed to build firmware on latest PlatformIO

25879, 2022-07-25:
I failed to build the firmware downloaded from here after i upgrad PlatformIO to the latest version (Core 6.1.3). 
It seems that the build-in library of ESP32 cannot be read correctly. 
Please help...
Part of the log attached below.. 
```
Found 38 compatible libraries
Scanning dependencies...
Dependency Graph
|-- TMCStepper @ 0.7.3
|   |-- SPI @ 1.0
|-- Adafruit MAX31865 library @ 1.1.0
|   |-- SPI @ 1.0
|-- Adafruit NeoPixel @ 1.5.0
|-- U8glib-HAL @ 0.4.1
|   |-- Wire @ 1.0.1
|   |-- SPI @ 1.0
|-- Arduino-L6470 @ 0.8.0
|-- SlowSoftI2CMaster @ 1.0.0
|-- AsyncTCP @ 1.1.1
|-- ESP Async WebServer @ 1.2.3
|   |-- AsyncTCP @ 1.1.1
|   |-- FS @ 1.0
|   |-- WiFi @ 1.0
|-- ESP3DLib @ 1.0.11+sha.5a2d484
|   |-- FS @ 1.0
|   |-- SPIFFS @ 1.0
|   |   |-- FS @ 1.0
|   |-- WiFi @ 1.0
|   |-- U8glib-HAL @ 0.4.1
|   |   |-- Wire @ 1.0.1
|   |   |-- SPI @ 1.0
|   |-- WebSockets @ 2.3.7+sha.dd8675c
|   |   |-- WiFi @ 1.0
|   |   |-- SPI @ 1.0
|   |-- ESP32SSDP @ 1.2.0+sha.6d78df5
|   |   |-- WiFi @ 1.0
|   |-- ESPmDNS @ 1.0
|   |   |-- WiFi @ 1.0
|   |-- Update @ 1.0
|   |-- ArduinoOTA @ 1.0
|   |   |-- Update @ 1.0
|   |   |-- WiFi @ 1.0
|   |   |-- ESPmDNS @ 1.0
|   |   |   |-- WiFi @ 1.0
|-- WebSockets @ 2.3.7+sha.dd8675c
|   |-- WiFi @ 1.0
|   |-- SPI @ 1.0
|-- ESP32SSDP @ 1.2.0+sha.6d78df5
|   |-- WiFi @ 1.0
|-- SPI @ 1.0
|-- EEPROM @ 1.0.3
|-- ArduinoOTA @ 1.0
|   |-- Update @ 1.0
|   |-- WiFi @ 1.0
|   |-- ESPmDNS @ 1.0
|   |   |-- WiFi @ 1.0
|-- ESPmDNS @ 1.0
|   |-- WiFi @ 1.0
|-- WiFi @ 1.0
|-- FS @ 1.0
|-- SPIFFS @ 1.0
|   |-- FS @ 1.0
|-- Wire @ 1.0.1
Building in release mode
Warning! We have just detected `upload_port` as IP address or host name of ESP device. `upload_protocol` is switched to `espota`.
Please specify `upload_protocol = espota` in `platformio.ini` project configuration file.
Compiling .pio\build\fysetc_e4\src\src\HAL\ESP32\FlushableHardwareSerial.cpp.o

... skip...

Compiling .pio\build\fysetc_e4\lib26e\arduinoWebSockets\WebSocketsClient.cpp.o
Compiling .pio\build\fysetc_e4\lib26e\arduinoWebSockets\WebSocketsServer.cpp.o
Compiling .pio\build\fysetc_e4\lib26e\arduinoWebSockets\libb64\cdecode.c.o
Compiling .pio\build\fysetc_e4\lib26e\arduinoWebSockets\libb64\cencode.c.o
Archiving .pio\build\fysetc_e4\lib148\libESPAsyncWebServer.a
In file included from .pio\libdeps\fysetc_e4\arduinoWebSockets\src\SocketIOclient.cpp:8:0:
.pio\libdeps\fysetc_e4\arduinoWebSockets\src\WebSockets.h:190:30: fatal error: WiFiClientSecure.h: No such file or directory

**************************************************************************
* Looking for WiFiClientSecure.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:WiFiClientSecure.h"
* Web  > https://registry.platformio.org/search?q=header:WiFiClientSecure.h
*
**************************************************************************

compilation terminated.
*** [.pio\build\fysetc_e4\lib26e\arduinoWebSockets\SocketIOclient.cpp.o] Error 1
In file included from .pio\libdeps\fysetc_e4\arduinoWebSockets\src\WebSockets.cpp:25:0:
.pio\libdeps\fysetc_e4\arduinoWebSockets\src\WebSockets.h:190:30: fatal error: WiFiClientSecure.h: No such file or directory

**************************************************************************
* Looking for WiFiClientSecure.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:WiFiClientSecure.h"
* Web  > https://registry.platformio.org/search?q=header:WiFiClientSecure.h
*
**************************************************************************

compilation terminated.
*** [.pio\build\fysetc_e4\lib26e\arduinoWebSockets\WebSockets.cpp.o] Error 1
In file included from .pio\libdeps\fysetc_e4\arduinoWebSockets\src\WebSocketsClient.cpp:25:0:
.pio\libdeps\fysetc_e4\arduinoWebSockets\src\WebSockets.h:190:30: fatal error: WiFiClientSecure.h: No such file or directory

**************************************************************************
* Looking for WiFiClientSecure.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:WiFiClientSecure.h"
* Web  > https://registry.platformio.org/search?q=header:WiFiClientSecure.h
*
**************************************************************************

compilation terminated.
*** [.pio\build\fysetc_e4\lib26e\arduinoWebSockets\WebSocketsClient.cpp.o] Error 1
In file included from .pio\libdeps\fysetc_e4\arduinoWebSockets\src\WebSocketsServer.cpp:25:0:
.pio\libdeps\fysetc_e4\arduinoWebSockets\src\WebSockets.h:190:30: fatal error: WiFiClientSecure.h: No such file or directory

**************************************************************************
* Looking for WiFiClientSecure.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:WiFiClientSecure.h"
* Web  > https://registry.platformio.org/search?q=header:WiFiClientSecure.h
*
**************************************************************************

compilation terminated.
*** [.pio\build\fysetc_e4\lib26e\arduinoWebSockets\WebSocketsServer.cpp.o] Error 1
======================================================================================== [FAILED] Took 447.78 seconds ========================================================================================

Environment    Status    Duration
-------------  --------  ------------
fysetc_e4      FAILED    00:07:27.785
==================================================================================== 1 failed, 0 succeeded in 00:07:27.785 ===================================================
=================================
```

brunoeiti, 2022-08-10:
Same problem here,  looking for a solution

cineast1, 2022-08-13:
Exactly same problem here, too

GerogeFu, 2022-08-19:
The firmware on the repo is pretty old one, seems have compatibility with latest platformio, anyone try latest marlin?

craftbot, 2022-08-20:
platformio is latest firmare from this github

craftbot, 2022-08-27:
Ie you need You need the following in Configuration_adv.h

  #define  X_SLAVE_ADDRESS 0
  #define  Y_SLAVE_ADDRESS 2
  #define  Z_SLAVE_ADDRESS 1
  #define E0_SLAVE_ADDRESS 3

in platformio.ini
[platformio]
src_dir      = Marlin
boards_dir   = buildroot/share/PlatformIO/boards
default_envs = fysetc_e4
include_dir  = Marlin

need config for your printer Configuration.h

Tigger777, 2022-09-14:
> The firmware on the repo is pretty old one, seems have compatibility with latest platformio, anyone try latest marlin?

I've got the latest Marlin 2.1.x to build and work as before.  It's not really that difficult if you follow Luc's instructions.

I used the default config files and modded them as follows:-

Configuration.h
============
1. Motherboard : BOARD_FYSETC_E4 2. 
2. #define SERIAL_PORT_2 -13. 
3. Change steppers drivers to TMC2209
4.Set Temp sensor 1 & bed to thermistor 1
5. Invert Stepper Directions to suit your setup.
6. Enable SDSUPPORT

Configuration_adv.h
===============
1. Uncomment  #define BABYSTEP_ZPROBE_OFFSET          // Combine M851 Z and Babystepping
2. Change sense resistor values from 0.11 to 0.22 - These were set to 0.11 in the fysetc repo here I'm not sure why - GeorgeFu do you know why?
3. Set the slave addresses of the TMS drivers as per craftbot's post above.

PlatformIO.ini 
==========
as per the above craftbot's post

I have also implemented a BL/CR touch on GPIO 0, but this requires a few more files to be changed.

4kardes, 2022-10-14:
same problem cant find solution

4kardes, 2022-10-16:
how about for 2.0.9
platform = espressif32@3.4
not working (compile but size too low and not work)
#define SERIAL_PORT_2 -1  getting error

Tigger777, 2022-10-17:
Sorry mate, that version doesn't compile anymore on my machine. Most probably from incessant updates to VSCode/Platform IO, you'd think they'd dial it back a bit now that they're supposed to be at war and all.  Try building 209 with ***@***.***, but you get wificlientsecure.h and a host of other errors each time you forcibly copy the files it's looking for into the project.

Juaneeto89, 2024-12-21:
Hello! I might be a little late to the party but I just got this same problem and found a way to solve it!

Basically, the problem is that  the board does not exist in PlatformIO. So what I did is I took another ESP32 based board like MKS TINYBEE, downloaded its firmware and started modifying the board file (json file in the board directory) to refer to the same pins that Fysetc E4 uses, I modified all the parameters to make it work on the same pinout and modified the configuration.h and configuration_adv.h according to what was refered in this thread. In other words, i preformed surgery on the firmware to make it believe it was being installed on a mks tinybee.

it worked. Let's hope you guys can get it to work, too.

suliaman99, 2025-05-17:
@Juaneeto89 
That is great, can you send your firmware file