FYSETC/FYSETC-E4 #42: Fix for WiFiClient library compilation fails again!

umartechboy, 2022-09-05:
Looks like it is supposed to have already been addresses in "Update platformio.ini to fix class WiFiClientSecure compilation error" {4b360345fc22ea1cb86909525ceb3c459a040e4e}, but for me it's still failing. Can't build the firmware!

Build terminal output: 
...
...
Compiling .pio\build\fysetc_e4\lib53d\arduinoWebSockets\WebSocketsClient.cpp.o
Compiling .pio\build\fysetc_e4\lib53d\arduinoWebSockets\WebSocketsServer.cpp.o
In file included from .pio/libdeps/fysetc_e4/arduinoWebSockets/src/WebSockets.cpp:25:0:
.pio/libdeps/fysetc_e4/arduinoWebSockets/src/WebSockets.h:192:30: fatal error: WiFiClientSecure.h: No such file or directory

craftbot, 2022-09-24:
need use marlin latest version

4kardes, 2022-10-07:
also not compile with new marlin 2.1.1

craftbot, 2024-02-03:
there should be no spaces on the path to the Marlin firmware