FYSETC/FYSETC-E4 #45: can't compile with new Platformio

4kardes, 2022-10-07:
cant compile with new platformio

getting error

Compiling .pio\build\fysetc_e4\liba94\arduinoWebSockets\WebSockets.cpp.o
In file included from .pio\libdeps\fysetc_e4\arduinoWebSockets\src\SocketIOclient.cpp:8:0:
.pio\libdeps\fysetc_e4\arduinoWebSockets\src\WebSockets.h:190:30: fatal error: WiFiClientSecure.h: No such file or directory

**************************************************************************
* Looking for WiFiClientSecure.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:WiFiClientSecure.h"
* Web  > https://registry.platformio.org/search?q=header:WiFiClientSecure.h
*
**************************************************************************

compilation terminated.
In file included from .pio\libdeps\fysetc_e4\arduinoWebSockets\src\WebSockets.cpp:25:0:
.pio\libdeps\fysetc_e4\arduinoWebSockets\src\WebSockets.h:190:30: fatal error: WiFiClientSecure.h: No such file or directory

**************************************************************************
* Looking for WiFiClientSecure.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:WiFiClientSecure.h"
* Web  > https://registry.platformio.org/search?q=header:WiFiClientSecure.h
*
**************************************************************************

compilation terminated.
*** [.pio\build\fysetc_e4\liba94\arduinoWebSockets\SocketIOclient.cpp.o] Error 1
*** [.pio\build\fysetc_e4\liba94\arduinoWebSockets\WebSockets.cpp.o] Error 1

Tigger777, 2022-10-14:
Have a look at Issue #40 the steps to follow are detailed there. Start with unmodified Marlin 2.1.x source, replace the  attached config files.

You'll need to edit /ini/esp32.ini 
edit the first line:-  platform = espressif32@3.4

[Configurations.zip](https://github.com/FYSETC/FYSETC-E4/files/9790220/Configurations.zip)

In the attached files I've added a comments "// !!!DO1409" wherever I made changes.

Good luck

4kardes, 2022-10-15:
thank you very much bro.
I solved the case because of you