FYSETC/FYSETC-E4 #51: Download mode not active

leviner18, 2023-01-18:
Hi,
Im trying to upload a sketch from Arduino IDE and i get following error:
A fatal error occurred: Failed to connect to ESP32: Wrong boot mode detected (0x13)! The chip needs to be in download mode.

Is there a possibility to put the board in download mode?

Thank you on advance

Br

apalazzi, 2023-04-01:
I have the same issue, did you find a way to solve it?

Jimbob-B, 2023-06-05:
Seems this can be fixed by adding a capacitor to the reset circuit - see the USB section on this page,

https://onstep.groups.io/g/main/wiki/32747

There's also a bit of a discussion here about similar issues and possible solutions,

https://github.com/espressif/esptool/issues/741

craftbot, 2024-02-03:
connect pin 25 on grnd

Rishi-00, 2024-07-13:
hey i tried this method in ESP-32-WROOM-S model but still thee issue persists, the code is not being upload
Please, help me as i am unable to get a solution
Thanks in advance