FYSETC/FYSETC-E4 #53: Support to LCD module?

RaulTrombin, 2023-08-04:


OriginaldCrypt, 2023-12-30:
The only feature to be the perfect board

TextileTopher, 2023-12-30:
> The only missing feature to be the perfect board

Agreed entirely. This was a really well designed and documented board. :-) There are just a few features I wish it had (a second UART or screen). 

I hope they do another revision of the board. I  B would even be willing to make some tutorials for it to help them out.

OriginaldCrypt, 2023-12-30:
> Agreed entirely. This was a really well designed and documented board. :-) There are just a few features I wish it had (a second UART or screen).

Does this mean that there is only one UART to control the TMC, and not available for a display? (having either the display or the TMC software control, not both at the same time)

TextileTopher, 2023-12-30:
> Does this mean that there is only one UART to control the TMC, and not available for a display? (having either the display or the TMC software control, not both at the same time)

From what I can see there are two. One used for the motor control chip and one used to program the device initially. I think you can repurpose that USB one, but you wouldn’t have serial output available then.

I am still experimenting. I really want to buy a whole bunch of these boards, but that extra uart is critical to me.

Tigger777, 2023-12-31:
The ESP32 doesn't have enough pins left for a screen on this board.  However you can add a gcode based screen that uses the serial port connected to the USB chip.   you can connect to the rx&tx lines on the motor uart header.  I the 4 pins closest to the ESP32 to power and communication.  I needed to add a 470r-1k resistor between esp RX and the screen TX.  This is quite an undertaking.  You need to rebuild marlin with all the reporting functions enabled.  Here's my one.  https://www.printables.com/model/323438-pendant-case-for-wt32-sc01, it served my purposes at the time.
I no longer use the E4 boards as I run klipper on my machines now. Actually I haven't used them since I got my Bambu X1C :)

OriginaldCrypt, 2023-12-31:
> you can add a gcode based screen that uses the serial port connected to the USB chip. you can connect to the rx&tx lines on the motor uart header. 

These two statements are contradictory, aren't they?
Do you have a drawing/sketch to clarify the design?

Thanks

Tigger777, 2023-12-31:
Ok let me clarify, the below the esp(if the antenna is the top) is labeled i2c on the one side and tx rx 5v g.  The centre row is is connected to all the tmc's the jumpers should be on the i2c side which is actually usart1 on the esp.  The other side is connected to usart0 and the usb chip via 1k series resistors. Sent from my phone

OriginaldCrypt, 2023-12-31:
Ok, I am afraid I don't really understood, but I got the point that TWO uart are present (not necesarilly available): one for the TMCs and the other one for the USB port. 

So, if I am not mistaken, we can make use of the USB to connect a what you call "GCODE based screen" but immediately after you say that it has to be connected to the "motor UART header".

Which is an example of such as "GCODE based screen"? WT32-SC01 ?
Which are the "motor UART header" pins? Won't it disable the motor UART control?

Thanks for your help!

Tigger777, 2023-12-31:
I called it the motor uart header because it allows you to jumper either usart0 or 1 to the tmc drivers. Yes your understanding is correct, the wtsc01 just overrides the usb chip signals and talks the controller like you would from a gcode  terminal. You could use any of the cheaper displays that are available now.Sent from my phone

Tigger777, 2024-02-04:
Sorry for the long delay in replying.  Here's the piece of the SCH file, 
[image]
and here's the PCB header P17-P18-P13 from left to right.
[image]

The jumpers are installed between P17 & P18. P13 is the TTL RS323 lines from the USB to serial chip.  there are 1k series resistors from the usb chip so you can assert signals on P13 to comm directly to the ESP chip and override the CH340's signals.

Tigger777, 2024-02-04:
Here's my board:
[image]

OriginaldCrypt, 2024-02-05:
It looks like the perfect workaround, do you have a direct link to purchase your exact model?

TextileTopher, 2024-02-05:
> Here's my board: [...]

Tiger - this is awesome. :-)

Tigger777, 2024-02-06:
No sorry I bought these boards Jan 2020, I had quick look on Aliexpress and they're all still v1.0 boards.  There isn't much you can change on these as there no more pins.  One change could be to use GPIO0 as the servo signal to drive a CR/BL touch which is what that green wire is on mine.  I'm happy to share the code for my screen, it's not the most elegant but it worked for me.  You will need to enable all the auto reporting functions in Marlin (temp, position, job progress).

OriginaldCrypt, 2024-12-06:
Hi all, I managed to get this working (at least, I managed to use a conection). I managed to do that after breaking a CYD and hours of testing, but finally it seems to be working. 

[image]

I use FluidNC (https://github.com/bdring/FluidNC) with FluidDial pendant (https://github.com/bdring/FluidDial). The pendant is built on a rotary enconder attached to a CYD (cheap yellow display) acting as remote controller (http://wiki.fluidnc.com/en/hardware/official/CYD_Dial_Pendant and M5 alterrnative http://wiki.fluidnc.com/en/hardware/official/M5Dial_Pendant).
FluidNC, for those who don't know, works like a Klipper firmware for 3d printers: a common firmware for the device and a configuration file, no need to recompile the software after a configuration change.

The Fysetc documentation is very bad. The documentation is incomplete. The electronic diagrams are incomplete. The pin references and pinouts are incomple/not named as in the diagrams. So, try and error. Some facts:

- Pins TX/RX on the right column (named P13 in the electronic diagrams) correspond to TX0/RX0 (as written in the back of the board). TX0 = GPIO1, RX0 = GPIO3. Those seems to be shared with the USB converter, but we can use them by referencing those GPIOs. 
[image]

the FluidNC configuration file excerpt for the UART is:

```
[...]
uart2:
  txd_pin: gpio.1
  rxd_pin: gpio.3
  rts_pin: NO_PIN
  cts_pin: NO_PIN
  baud: 1000000
  mode: 8N1

uart_channel2:
  report_interval_ms: 75
  uart_num: 2
[...]
```

- I will test more later , but it looks like the UART0 can be used along with UART1 (motor UART) but not with USB.

BR